berry-lang/berry #373: %q format specifier is invalid

Alex-x90, 2023-10-14:
According to [Chapter-7.rst](https://github.com/berry-lang/berry/blob/2785c08157d07bfad175f1ceebeb9716db652518/docs/source/en/Chapter-7.rst), there should be a format specifier %q which results in the string being between quotes with special characters escaped (equivalent to string.escape as I understand). When I run the following code it says `invalid option '%q' to 'format'`.
```
import string
print(string.format("%q","test"))
```

s-hadinger, 2023-10-14:
This is a documentation bug. It was never implemented. You should use %q with `json.dump()` as a replacement. 

I will check however if it's easy to add

sfromis, 2023-11-11:
On 2023-07-09 I removed the unimplemented `%q` from the wiki markdown page [Chapter 7](https://github.com/berry-lang/berry/wiki/Chapter-7), which seems to be where the most recent docs are, instead of the `.rst` files in `docs/source/en`.

s-hadinger, 2024-04-17:
%q is now supported, short manual and documentation have been updated

closed 2024-04-17 by s-hadinger