berry-lang/berry #375: class forward call definition problem

paddy0489, 2023-10-18:
berry.exe state.be is working well 
but if i call berry and import state.be 
than an redefinition error comes.

> import "state.be" as state
syntax_error: state.be:17: redefinition of 'OffState'
stack traceback:
        stdin:1: in function `main`

```
#- Step 1: Define the State Interface
class State()
    def handle(context):
        pass
    end
end
-#
# Step 2: Create Concrete States
var OffState = nil
class OnState
    def handle(context)
        print("Light is already ON. Switching it OFF now.")
        context.state = OffState()
    end
end

class OffState
    def handle(context)
        print("Light is OFF. Switching it ON now.")
        context.state = OnState()
    end
end

# Step 3: Create the Context
class LightSwitch
    var state
    def init()
        self.state = OffState()
    end
    def press()
        self.state.handle(self)
    end
end

# Testing the FSM
switch = LightSwitch()
switch.press()  # Expected: "Light is OFF. Switching it ON now."
switch.press()  # Expected: "Light is already ON. Switching it OFF now."
switch.press()  # Expected: "Light is OFF. Switching it ON now."
```

s-hadinger, 2023-10-18:
Hmm. I don't see an easy way around this. Let's take a simplified example:

```
var B = nil
class A def init() return B end end
class B def init() return A end end
```

But when you import it, Berry implicitly wraps it in a virtual function equivalent to:
```
def f()
    var B = nil
    class A def init() return B end end
    class B def init() return A end end
end
```

In the implicit function, you can't redefine a variable with a class. Unfortunately the only way I can think about would be to use indirection variables (I put 2 of them for symmetry):

```
var A = nil
var B = nil
class A_class def init() return B end end
class B_class def init() return A end end
A = A_class
B = B_class
```

paddy0489, 2023-10-18:
It's working, thank you very much :-)

It looks like a workaround, will it be fixed in a future interpreter version?

s-hadinger, 2023-10-19:
I'm not sure we can easily fix this. This is the price to pay for a lightweight single pass compiler.

closed 2025-11-03 by s-hadinger