berry-lang/berry #385: Documentation sync

s-hadinger, 2023-12-25:
Hi all. Thanks for the effort done on the documentation here: https://berry.readthedocs.io/en/latest/

However I just realized that the latest changes to the wiki were not reflected in the documentation web site. This is bad.

Why do we need to copy from the wiki to the main repo? Couldn't we just use the wiki to convert to readthedocs?

diazvictor, 2024-04-15:
> Hi all. Thanks for the effort done on the documentation here: https://berry.readthedocs.io/en/latest/
>
> However I just realized that the latest changes to the wiki were not reflected in the [...]

We can make a link and place it on the web.

s-hadinger, 2024-04-15:
I don't know who owns https://berry.readthedocs.io/en/latest/

It seems to be @hiperiondev but he doesn't respond anymore

diazvictor, 2024-04-16:
> I don't know who owns https://berry.readthedocs.io/en/latest/
> 
> It seems to be @hiperiondev but he doesn't respond anymore

I thought it came from https://github.com/berry-lang/berry_doc

s-hadinger, 2024-04-16:
Nope, it doesn't. It comes from https://github.com/hiperiondev/berry which we have no control on.

diazvictor, 2024-04-17:
> Nope, it doesn't. It comes from https://github.com/hiperiondev/berry which we have no control on.

We have no control over it because it was supposed to be set up here in the organization, but I don't know what happened afterwards, I had a conversation with him in the telegram group 2 years ago about it.

s-hadinger, 2024-04-17:
Yes please. Because now we are stuck with a documentation outdated and frozen in time; which even worse than before. I understand he is not motivated anymore, and I would be grateful he passes back control to us.

s-hadinger, 2024-04-24:
@diazvictor Did you managed to contact @hiperiondev ?

s-hadinger, 2024-05-14:
I finally took control back of https://berry.readthedocs.io/en/latest/ and updated it with the latest additions to the wiki. The source is now https://github.com/berry-lang/berry_doc

There will be more polishing in the coming days

closed 2024-05-14 by s-hadinger