berry-lang/berry #395: Loading / importing an external script

max22-, 2024-02-06:
Hello,
i've searched through the documentation but i haven't found a way to import a script from another one.
what is the best way to do it ?
the current workaround i am using is this :
```
class MyClass
end
compile('myclass.be', 'file')()
```
I have to create an empty class with the same name as the one i am importing otherwise i get a syntax error.
Is there an equivalent of python 'import' that i am missing ?

Thank you,
Maxime

s-hadinger, 2024-02-06:
There are basically two ways:
- `compile()` like above, that compiles a source code in the current context
- `import` (preferred) that loads the file with the same name, and stores in the imported name what the source file returns (typically a module). The result of import is cached.

See here: https://github.com/berry-lang/berry/wiki/Chapter-8#83-how-to-package-a-module

max22-, 2024-02-06:
ok ! thank you very much !