berry-lang/berry #396: ternary assign to local variable bug

JayXon, 2024-02-19:
```berry
if true
  a=1
  a=true ? a+1 : a+2
  print(a)
end
```
this prints 1

s-hadinger, 2024-02-19:
I will look at it next week

s-hadinger, 2024-02-21:
I confirm the bug. Compiling the following code:

```berry
def f()
  if true
    var a = 1
    a = true ? a+1 : a+2
    print(a)
  end
end
```

then doing `import solidify solidify.dump(f)` gives the following:

```C

    ( &(const bvalue[ 2]) {     /* constants */
    /* K0   */  be_const_int(1),
    /* K1   */  be_const_int(2),
    }),
    &be_const_str_f,
    &be_const_str_solidified,
    ( &(const binstruction[12]) {  /* code */
      0x50000200,  //  0000  LDBOOL	R0	1	0
      0x78020008,  //  0001  JMPF	R0	#000B
      0x58000000,  //  0002  LDCONST	R0	K0               <- var a is in R0
      0x50040200,  //  0003  LDBOOL	R1	1	0
      0x78060001,  //  0004  JMPF	R1	#0007
      0x00040100,  //  0005  ADD	R1	R0	K0       <- ERROR: the result of a+1 is stored in R1
      0x70020000,  //  0006  JMP		#0008
      0x00000101,  //  0007  ADD	R0	R0	K1       <- CORRECT: the result of a+2 is stored in R0
      0x60040001,  //  0008  GETGBL	R1	G1
      0x5C080000,  //  0009  MOVE	R2	R0
      0x7C040200,  //  000A  CALL	R1	1
      0x80000000,  //  000B  RET	0
    })
```

s-hadinger, 2024-02-29:
It's fixed in https://github.com/berry-lang/berry/pull/404

Thanks for reporting

PR #404 (2024-03-24): Fix ternary parser bug
Fix #396

Details:

The Berry parser does simple optimization in the following pattern:
```
def f() var a = 1 a = a + 2 end
```

The raw bytecode generated is:
```
0000  LDCONST	R0	K0
0001  ADD	R1	R0	K1
0002  MOVE	R0	R1
0002  RET	0
```
but the optimizer saves the `MOVE`:
```
0001  ADD	R1	R0	K1
0002  MOVE	R0	R1
# replaced by
0001  ADD	R0	R0	K1
```
storing directly the result in R0 instead of storing into a temporary R1 that is discarded right after.

In the case of the ternary operator, only the last `ADD` is patched, not the first one.

Fix: I added a `lastjmp` informati [... cut]
closed 2024-02-29 by s-hadinger