berry-lang/berry #410: Build fails in Windows using cmake

yuppox, 2024-04-11:
Hi.

I'm trying to test this out in Windows. I am using the MS build tools with cmake.

`berry-coc` builds and everything else fails.

The first issue I see repeating is a problem with "unknown sizes".

```
C2057: expected constant expression
C2466: cannot allocate an array of constant size 0
C2133: 'id_buf': unknown size
```

How can I get things working?

s-hadinger, 2024-04-11:
This looks like a C version issue. I don't use the MS, I prefer to use `w64devkit` which compiles fine.

yuppox, 2024-04-11:
Yes, it seems to do with the C version supported by the MS tools.

yuppox, 2024-04-11:
Unless code changes are implemented, it's not going to be possible. Good to know.

closed 2024-04-11 by yuppox