berry-lang/berry #424: Out-of-bound memory access in be_byteslib

Kosehai, 2024-05-22:
Due to a cast from size_t to int32 in buf_set and buf_get functions it is possible to supply a large number to access memory out-of-bounds from a bytes object.

Here is an example problematic function:
```c
static void buf_set4_le(buf_impl* attr, size_t offset, uint32_t data)
{
    // Cast here is causeing the problem
    if ((int32_t)offset + 3 < attr->len) {
        attr->bufptr[offset] = data & 0xFF;
        attr->bufptr[offset+1] = (data >> 8) & 0xFF;
        attr->bufptr[offset+2] = (data >> 16) & 0xFF;
        attr->bufptr[offset+3] = data >> 24;
    }
}
```

### Reproducing

Here is a berry code snippet that will cause a segmentation fault:
```lua
b=bytes()
b.set(32314545516,0,4) #This is being treated as a negative number bypassing the bounds check
```

### Fixing

The issue can be rectified by checking that the provided offset is not negative.

s-hadinger, 2024-05-27:
Nice finding. Maybe we should handle negative offset as starting from end, like in arrays. Thoughts?

s-hadinger, 2024-07-24:
Fixed, awaiting approval before merging

PR #440 (2024-08-23): Bytes accept negative offset
Bytes accepts negative offset, that are considered as positions from the end of the buffer.

If the offset is out of range, reads return zero, and writes raise an exception.

Fixes #424
closed 2024-08-23 by skiars