berry-lang/berry #431: Proposal: add arguments to `int()` to guard value and add default

s-hadinger, 2024-06-15:
`int()` currently takes only one argument, and return `nil` if the argument is `nil`

I'm facing a lot of code where I need to guard the value within a range and potentially define a default value.

```berry
if (i == nil)  i = 0 end
if (i < 0) i = 0 end
if (i > 100) i = 100 end
```

I'm proposing to have new optional arguments:
`int(arg:any, [min: int, max: int, def: int or nil]) -> int or nil`

With this the above code would look like:
```berry
i = int(i, 0, 100, 0)
```

It could also used without a default value, hence forcing an `int` between `0` and `100` or `nil`

```berry
int(-1, 0, 100)     # 0
int(1, 0, 100)      # 1
int(101, 0, 100)    # 100
int(2.5, 0, 100)    # 2
int(nil, 0, 100)    # nil
```

@skiars any opinion on this proposal?

s-hadinger, 2024-06-27:
@skiars any opinion for or against it? If not I will implement it as described above.

skiars, 2024-06-28:
I'm a bit confused about the use of `max`, but not `default`. I suggest naming this argument `default` instead of `def`, because `def` is a keyword.

s-hadinger, 2024-06-28:
Thanks, using a default value was indeed a bad idea and makes the syntax confusing.

I agree that min/max are missing.

But then the syntax would be:
```
v = math.bound(0, int(v), 10)
```

Last point, should it be `bound(min, val, max)` or `bound(val, min, max)`?

skiars, 2024-06-29:
The `v = math.bound(0, int(v), 10)` is pretty readable. As for the default value of `int()`, let's take a look at other languages:

- `parseInt()` in JavaScript returns `NaN` by default (I don't like `NaN`, it can easily break calculations)
- In Python, the default constructor of `int()` is `0`, but it may raise an error during parsing