berry-lang/berry #432: SUGGESTION: Comparison with other languages

yuppox, 2024-06-18:
Hi,

As a suggestion, I would like to see a comparison of Berry and other languages. I assume that would focus on scripting languages. For example:

JavaScript
Julia
Lua
Perl
PHP
Python
Ruby
Squirrel

It would be nice to know more about what advantages Berry has and why it's unique/different.

Thanks!

s-hadinger, 2024-06-19:
It depends on what you are looking for. Berry's primary goal is to consume as little memory as possible to run on embedded system, and having a small code footprint as well. Berry still provides a python-like syntax (more or less), classes and closures, making it powerful enough for many embedded use cases. For example, Berry has a single pass compiler: it keeps compiler code small and has low memory usage (no AST in memory), but also limits ability to do compile-time optimizations.

As such it does not really compare to PHP or Perl. Alternatives are Lua (whom syntax you may not like as much) or MicroPython (which cannot be used as a simple library contrary to Berry).

You can find here a simple syntax comparison for Python programmers: https://berry.readthedocs.io/en/latest/source/en/Berry-in-20-minutes.html#extra

yuppox, 2024-06-19:
Thanks for the reply. For me, small for embedded sounds good.

I still think it would be helpful to put together a page in the documentation with some common comparisons. Many languages wouldn't be relevant but there are quite a few scripting languages (e.g. https://github.com/dbohdan/embedded-scripting-languages). There would be appropriate comparisons and it is useful explore the differences.

I like Lua and use it frequently. However, it seems to me that Berry has better support for objects. Lua is perhaps more flexible in that area but I think people generally prefer straight-forward. I'm also not sure I like the 1-based index.

I dislike the Python syntax and much prefer Berry. I'm sure Berry is also considerably faster than Python for many tasks. And as you mention, Berry has advantages over MicroPython.

So, that's my suggestion. If anyone wants to share comparisons, I think it's a good exercise.

Also, I think Berry should be added to that list (https://github.com/dbohdan/embedded-scripting-languages)!

sfromis, 2024-06-21:
While this is not a comparison, you can still check the coding examples, and do your own mental matching to how you'd do in other languages you know.
https://github.com/berry-lang/berry/tree/master/examples

As Berry does not have much in unusual features or cryptic operators, reading the code should be pretty straightforward, if you know other languages, especially Python.

skiars, 2024-06-28:
If you hope to use popular scripting languages on the MCU, I recommend QuickJS, but it is much larger than the Berry interpreter (RAM and ROM). In contrast, Berry’s syntax design considers simplifying the interpreter design, without sacrificing too much performance.

If you are familiar with popular scripting languages such as JavaScript or Python, you can also quickly get started with Berry. If you are not planning to run existing JavaScript or Python code on the MCU, I don’t think there is a problem with writing new Berry code.

sfromis, 2024-06-28:
As things have become now, it looks like people wanting to get started with some language (or whatever) are increasingly turning to ChatGPT (or another LLM), which struggles to generate working Berry code, but doing much better with the most popular scripting languages.

Best case, you may get something which is "close", but you still have to know the language to work out where the model hallucinated syntax details. I just did a small test with a Quicksort algorithm, and after fixing some syntax and semantics errors, the basic structure from ChatGPT was workable.

skiars, 2024-06-29:
> As things have become now, it looks like people wanting to get started with some language (or whatever) are increasingly turning to ChatGPT (or another LLM), which struggles to generate working Berry code, but doing much better with the most popular scripting languages.

LLM is really good at generating Python code (they always generate Python code, even if a library is written in C, they will tend to generate the Python binding for it). However, perhaps we could make a fine-tuning dataset in the future to help LLM generate better Berry code.

sfromis, 2024-07-13:
> Also, I think Berry should be added to that list (https://github.com/dbohdan/embedded-scripting-languages)!

It is now :smile:

mi-hol, 2024-10-15:
> It would be nice to know more about what advantages Berry has and why it's unique/different.

I had the same question and was able to answer it quickly via an internet search using the search term:
"Berry versus LUA" 
one of the best hits was https://www.libhunt.com/compare-berry-lang--berry-vs-lua